TinyCircuits/TinyCircuits-Thumby-Code-Editor #2: Multiplayer support?

marvbloke, 2021-09-29:
As the Thumby has 2 player multiplayer support via a micro USB to micro USB cable, would be great if at least one of the bundled games supported multiplayer. 2 player Tetris on the Game Boy is always great fun, so would love if TinyTris had the option with the same sort of rules (e.g. complete sets of lines to send lines to your opponent - 2 lines sends 1, 3 lines sends 2, 4 lines sends 4). Win if you're the first to complete 30 lines or opponent goes off the top of the screen.

ace-dent, 2021-09-29:
Pong was suggested by 'Coolie Coolster' in KS comments.

TinyCircuits, 2021-09-29:
We have plans in the works to show off multiplayer support - stay tuned on the Kickstarter for updates coming soon!