TinyCircuits/TinyCircuits-Thumby-Code-Editor #3: Bitmap formatting for readability

ace-dent, 2021-09-29:
It would be great if the output bitmap 'array' (tuple) could be formatted to visually indicate the data structure / image.
e.g.

```python
# BITMAP: width: 8, height: 8
bitmapHero = (
    0,  # ▓▓▓▓▓▓▓▓
    0,  # ▓▓▓▓▓▓▓▓
   16,  # ▓▓▓░▓▓▓▓
  150,  # ░▓▓░▓░░▓
  254,  # ░░░░░░░▓
    0,  # ▓▓▓▓▓▓▓▓
    0,  # ▓▓▓▓▓▓▓▓
    0   # ▓▓▓▓▓▓▓▓
 )
```

I appreciate this works best for small (height 8px /1byte or less) images. Perhaps add it as an option for 'verbose' / 'detailed' output?

ace-dent, 2021-10-20:
@raymond-li
Nice idea. I've experimented previously with the braille unicode hack, but find it's too inconsistently displayed. Often the spaces between braille character blocks make the overall image incongruous. In my experience, the simpler byte representation with blocks is more consistent across fonts and platforms.
 e.g. https://github.com/filmote/Font4x6/blob/519060bc6f321c2a0c511e4e415cc55acb1c31c1/src/fonts/Font4x6.cpp#L33

raymond-li, 2021-10-21:
@ace-dent , yeah the braille doesn't look great, especially for smaller sprites.

The issue I see with the inline blocks your describe is that it doesn't really work inline if the thumby sprite data is taller than 8 pixels, since each byte of the sprite data represents up to 8 rows of pixels. The other downside, to me, is that the inline block representation means the image is rotated.

If you forget about trying to put the preview inline, you can just have it printed before the sprite, which looks the best to me, but that could be unwieldy for large sprites and doubles the number of lines, which I wouldn't be that opposed to, personally. For larger projects, you'd probably want to put your sprites into a separate file anyways, although the Web IDE doesn't support multi-file projects yet, I don't think. Example:
```python
# ▓▓▓▓▓▓▓▓
# ▓▓▓░░▓▓▓
# ▓▓▓░░▓▓▓
# ▓▓▓▓░▓▓▓
# ▓▓░░░▓▓▓
# ▓▓▓▓░▓▓▓
# ▓▓▓▓░▓▓▓
# ▓▓▓░░▓▓▓
# BITMAP: width: 8, height: 8
bitmapHero = (
    0,
    0,
   16,
  150,
  254,
    0,
    0,
    0 
)
```

Another problem is that the Web IDE/Emulator doesn't like these unicode characters in the comments. I get this:
```
import thumby

print("This script fails because of the unicode comment below")

# ▓▓▓▓▓▓▓▓ 

print("This line fails because of the unicode comment above")
```
```
import main
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 7
SyntaxError: invalid syntax
Emulator stopped
```
Maybe there's a way to work around it by putting a specific character after the unicode comment. Not sure.

ace-dent, 2022-12-11:
I'm not sure how relevant this feature request is... general python style is for quite compact code.
Marking as Closed. Feel free to reopen.

jmtinycircuits, 2023-03-02:
I don't think we'll ever do this, unfortunately. Although cool, too much work to implement and it could cause additional editor lag.

See this page for an example of an editor that does this but causes lag: [page](https://corax89.github.io/esp8266Game/index.html?src=398002b70a9a35970810b355d44eb941)

closed 2023-03-02 by jmtinycircuits