TinyCircuits/TinyCircuits-Thumby-Code-Editor #4: IDE Feature Requests

raymond-li, 2021-10-17:
Thank you for releasing the beta IDE, emulator, and examples! It's already very fun to play around with.

I had a few suggestions/requests that you've probably already thought about, but I wanted to voice my personal wish list in hopes it pushes them up a little :smile: 

- Emulator console for debugging and feedback. As I wrote this I just found out the messages get printed to the Chrome console. I really wish I knew that earlier! I think this would make sense exposed as another "Shell" window.
- Instructions for offline development with the emulator. It would be handy to be able to develop and test from my own editor as the autocomplete/intellisense is better. A VSCode extension with the emulator, pixel editor and Shell would be awesome but probably a pipedream.
- Emulation of the Filesystem and Shell views. If we could develop through the emulator with everything looking the same as if a Thumby were plugged in, that would be pretty cool.
- Better system for sharing games. Right now to try out someone's code I have to download their code, copy/paste it into the editor (potentially overwriting an existing project), and then hit emulate. It would be neat if at least the first two steps were simplified. Hosting files probably isn't attractive, so maybe linking data from Github Gists or files could work?
- More Thumby/Micropython documentation. Most answers can be found in the Thumby source code, examples, or Micropython source code if you know how to look. Some things, mostly with regards to capabilities, are less obvious. For example, I was looking for what characters can be rendered with `thumby.display.drawText()` but didn't find an answer. I assume all printable ASCII characters are supported at least, but I haven't checked.

Sorry if Issues isn't the right channel for this kind of discussion. It seemed a little more appropriate than the forum.

jmtinycircuits, 2021-10-19:
Hi,

1. We are planning on having a shell attached to the emulator, or maybe integrating the emulator into its own panel (like the bitmap builder) and outputting to the existing hardware shell. This should be coming in the near future.
2. A VSCode extension would be cool (it's what I use for most stuff) but unfortunately will have to fall to the back of the list of things to look into. Just in case, files can be imported from a computer to the web editor through the File -> import file from PC button, but I know this is not an ideal/fast workflow.
3. The emulator libraries used do not support filesystem mounting out of the box in a browser. One way I see this working is if all the files are defined before starting the emulator, then it's possible to write most of filesystem to the emulator flash.
4. We are looking into various ways to do this. Gists could work, this is one solution I plan on trying to implement soon.
5. We have Thumby API documentation now: https://github.com/TinyCircuits/tinycircuits.github.io/blob/master/ThumbyAPI.md let us know if anything can be more clear

Thanks for the feedback!

jmtinycircuits, 2021-10-20:
> Yeah, an extension would be pretty involved and hard to justify working on. I'd be interested to hear a little bit about how the emulator stack works, but it sounds like the emulator is still being worked on.

The underlying emulator library is still under development: https://github.com/wokwi/rp2040js. There are some other libraries the by the same author like https://libraries.io/npm/littlefs and the uf2 formatter https://www.npmjs.com/package/uf2 that are used. The relevant files for the web IDE are 

- https://github.com/TinyCircuits/tinycircuits.github.io/blob/master/js/emulator_wrapper.js
- https://github.com/TinyCircuits/tinycircuits.github.io/blob/master/js/load-file.js
- https://github.com/TinyCircuits/tinycircuits.github.io/blob/master/js/load-uf2.js

> If you implement it like that, would that mean any code that writes to the file system will fail?

Just looked, yes it fails to write data to the pre-written files on the emulated flash. Looks the way I said I might do means files would still be read only.

> I don't think the audio functions work yet in the emulator, do they? Maybe a quick blurb like was done for the "Files" section would be helpful if that's the case:

Yeah, I'll do that as well as update that file writing doesn't work right now as well.