TinyCircuits/TinyCircuits-Thumby-Code-Editor #8: Degrade gracefully for unsupported browsers

ace-dent, 2021-10-30:
The readme makes it clear that only specific browsers are supported (due to current adoption of WebSerial).
However, many casual users will just try to use the Web IDE in their default (unsupported) browser- which just makes things appear... broken...

Please degrade gracefully. After doing a browser check, warn the user about compatibility issue. 
Consider allowing certain functions to still be useable- such as the bitmap editor.

jmtinycircuits, 2021-11-01:
This does happen to some level now.
[image]
Although we do not go through and try to ensure everything works on the unsupported browsers.

ace-dent, 2021-11-02:
Doesn't work for me with macOS Safari :-(

jmtinycircuits, 2021-11-04:
OK, I'll try to implement a check for the Safari browser in the next version of the IDE.

jmtinycircuits, 2023-09-28:
Is this still an issue today? Detecting which browser on which OS is a whole can of worms. It's weird that WebSerial is detected in Safari.

The detection happens here: https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/blob/430a4b2b5b5e03d68fe832d050ff902bcad05482/js/repl.js#L55-L60