TinyCircuits/TinyCircuits-Thumby-Code-Editor #11: Google Chrome 81.0.4044.92 + Linux not working

thor27, 2022-01-21:
Hello,
I'm trying to open https://tinycircuits.github.io/ on Google Chrome version 81.0.4044.92  on my Ubuntu machine but it's not working, I just get a gray screen. On the console is printed this error:
```
Uncaught SyntaxError: Unexpected reserved word
```

Screenshot
[image]

Already tried to fully clean local storage, cache and cookies and no success

jmtinycircuits, 2022-01-21:
Simple enough to fix. Will close on next merge.

jmtinycircuits, 2022-01-21:
Fixed page is active, fetch it with ctrl+F5. Let me know

thor27, 2022-01-21:
Hi,

Thanks for the quick reply and fix! Now I'm getting few other issues:

1. When I open I get:
```
Serial NOT supported in your browser! Use Microsoft Edge or Google Chrome
```

Even though I'm using an official google chrome build

2. Emulator has some visual issues:
[image]

3. Connect thumby does nothing

thor27, 2022-01-21:
Hi, It seems the auto update of the browser was broken. I manually updated to the latest version and it's working now.

Thanks!

closed 2022-01-21 by thor27