TinyCircuits/TinyCircuits-Thumby-Code-Editor #12: Microsoft Edge 99.0.1135.6 + Linux not accessing thumby file system

thor27, 2022-01-21:
Hi,

On MIcrosoft Edge version 99.0.1135.6 running on Ubuntu, I'm able to connect to thumby successfully but I can't see the filesystem or "upload as" the demo code:

[image]

Emulator works fine.

jmtinycircuits, 2022-01-21:
It looks like the Thumby is not connecting all the way. You should get a REPL that you can type commands into. Disconnect it, power cycle it, and reconnect.

Look at the shell in my image.
[image]

Make sure you don't have any other IDE tabs open.

thor27, 2022-01-21:
yes, it worked! strange that I tried exactly the same thing before and didn't worked that is why I posted the issue. But anyway, problem fixed :) thanks!

closed 2022-01-21 by thor27