TinyCircuits/TinyCircuits-Thumby-Code-Editor #19: Support for not qwerty keyboard layout

ultrasuperpingu, 2022-03-21:
The emulator map the button with wasd for the dpad and , and .for B and A
It would be nice to be able to change those. On an azerty keyboard for example, period need the shift to be pressed, and the equivalent of wasd would be zqsd.

ace-dent, 2023-03-15:
Adding cursor key support would be nice (secondary key bindings).
<kbd>▲</kbd> <kbd>▼</kbd> <kbd>◄</kbd> <kbd>►</kbd>
But then what would be a safe choice `A` and `B`? Perhaps <kbd>1</kbd> & <kbd>2</kbd> is universal ?

jmtinycircuits, 2023-03-15:
I way to reassign key binds would likely make the most people happy.

ace-dent, 2023-05-26:
Also, noted in Discord: Currently key presses are case sensitive. 
Users expect caps lock to not affect key press detection.
(https://discord.com/channels/898292107289190461/898292174410612787/1111446002784796762)