TinyCircuits/TinyCircuits-Thumby-Code-Editor #20: Garbage collector does not seem to work

ultrasuperpingu, 2022-03-21:
I tried this simple script:
```
import gc
import micropython

gc.collect()
micropython.mem_info()
print('-----------------------------')
print('Initial free: {} allocated: {}'.format(gc.mem_free(), gc.mem_alloc()))
def func():
    a = bytearray(10000)
    del(a)
gc.collect()
print('Func definition: {} allocated: {}'.format(gc.mem_free(), gc.mem_alloc()))
func()
print('Func run free: {} allocated: {}'.format(gc.mem_free(), gc.mem_alloc()))
gc.collect()
print('Garbage collect free: {} allocated: {}'.format(gc.mem_free(), gc.mem_alloc()))
print('-----------------------------')

micropython.mem_info(1)
```
But the result in emulator console was:
```
stack: 1140 out of 7936
GC: total: 192064, used: 6224, free: 185840
 No. of 1-blocks: 40, 2-blocks: 21, max blk sz: 72, max free sz: 11455
-----------------------------
Initial free: 185840 allocated: 6224
Func definition: 185808 allocated: 6256
Func run free: 175664 allocated: 16400
Garbage collect free: 175808 allocated: 16256
-----------------------------
stack: 1140 out of 7936
GC: total: 192064, used: 16384, free: 175680
 No. of 1-blocks: 46, 2-blocks: 20, max blk sz: 625, max free sz: 10830
GC memory layout; from 20007e60:
00000: h=MhhhDShS..hDhS.DBBBh===DBDh====B=BBBBBBBB=B.B=BBB.B=.B.B=Bh===
...
```
The allowed 10kB are not collected, event with a del command.
Maybe I missed something. If so, I am really sorry.

jmtinycircuits, 2022-03-21:
I get similar results on the hardware.

```
stack: 1068 out of 7936
GC: total: 192064, used: 6848, free: 185216
 No. of 1-blocks: 54, 2-blocks: 20, max blk sz: 65, max free sz: 11184
-----------------------------
Initial free: 185216 allocated: 6848
Func definition: 185184 allocated: 6880
Func run free: 175040 allocated: 17024
Garbage collect free: 175184 allocated: 16880
-----------------------------
stack: 1068 out of 7936
GC: total: 192064, used: 17008, free: 175056
 No. of 1-blocks: 60, 2-blocks: 19, max blk sz: 625, max free sz: 10559
GC memory layout; from 20008040:
```

Garbage collection is weird in Python/MicroPython.

Without the function the `bytearray` gets collected as intended:
```
>>> import gc
>>> import micropython
>>>
>>> gc.collect()
>>> micropython.mem_info()
stack: 476 out of 7936
GC: total: 192064, used: 44000, free: 148064
 No. of 1-blocks: 271, 2-blocks: 48, max blk sz: 112, max free sz: 4845
>>>
>>> a = bytearray(10000)
>>> micropython.mem_info()
stack: 476 out of 7936
GC: total: 192064, used: 54208, free: 137856
 No. of 1-blocks: 276, 2-blocks: 52, max blk sz: 625, max free sz: 4845
>>>
>>> del a
>>> gc.collect()
>>> micropython.mem_info()
stack: 476 out of 7936
GC: total: 192064, used: 43984, free: 148080
 No. of 1-blocks: 272, 2-blocks: 47, max blk sz: 112, max free sz: 4845
```

This has to do with Python/MicroPython rather than the Thumby/Emulator

ultrasuperpingu, 2022-03-21:
You're right, it seems ok when de variable is affected in the global scope. But affecting a global variable in the function does not work either...
That's weird because I found this code in the micropython docs (https://docs.micropython.org/en/latest/reference/constrained.html#reporting) and it says:
`Running the function uses over 10KiB, but on return a is garbage because it is out of scope and cannot be referenced. The final gc.collect() recovers that memory.`

I'll try to check if an issue about this is present in micropython's project. What version of micropython do you use?
Thanks anyway for the reply.

jmtinycircuits, 2022-03-21:
We're currently using v1.18

closed 2022-03-28 by jmtinycircuits