TinyCircuits/TinyCircuits-Thumby-Code-Editor #47: Integrate grayscale support

aroidzap, 2022-12-25:
Please integrate https://github.com/Timendus/thumby-grayscale into official Thumby lib.

masonova1, 2023-01-09:
The community is having integration problems with newer revisions. Even if the newer revisions did allow for it, the features present in the grayscale library are the product of some very delicate and hardware-specific tunings that we can't guarantee uniform support for over time, and it would be a little unwieldy for the typical user to dial it in manually.

closed 2023-01-09 by masonova1

aroidzap, 2023-01-09:
Couldn't we just add some "experimental grayscale" setting in settings? Every app using grayscale could fallback to regular B/W.

masonova1, 2023-01-10:
If art assets were made to be displayed in grayscale, and grayscale isn't supported, I'm not sure it's sufficient for many artists to just quantize the grayscale sprites -- I'm worried that doing this would practically mean asking many grayscale game developers to make another set of B/W assets, just in case a particular Thumby won't support grayscale.