TinyCircuits/TinyCircuits-Thumby-Code-Editor #50: Possible API typo: HWID

ace-dent, 2022-12-29:
When `thumby.py` imports `thumbyHardware.py`, it exposes `IDPin`.
This is likely a typo and should be `HWID`. There is no reason to make IDPin accessible through the API.
Affects both `lib` and `lib-emulator` versions.

ace-dent, 2022-12-29:
Follow-up question: if HWID is available, should `i2c` and `spi` be accessible through the API? Why is it required?Should this be part of `thumbyGraphics.py` where initiation is self-contained using `HWID` ([here](https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/blob/6c2c18af087254be715221fa34ec505e285cdba9/ThumbyGames/lib/thumbyGraphics.py#L382-L387)).

masonova1, 2023-01-09:
None of the software we've ever seen actually uses the ID pin, so I'm fine with deprecating it. We just need to de-duplicate HWID-getting code inside of `main` from `thumbyHardware`, and then we can get the HWID straight from `thumbyHardware` and remove the import. I'll do this soon.