TinyCircuits/TinyCircuits-Thumby-Code-Editor #52: OLED GDDRAM is not cleared on initialisation

ace-dent, 2022-12-31:
### Problem:  
When working with a customised `main.py`, we cannot rely on the screen buffer overwriting the OLED onboard 'Graphic Display Data RAM' (GDDRAM). This can lead to the unexpected result of showing random data on first boot, or restoring the last image after a reset.
Data persists after power cycling On `0xAF` / Off (sleep screen) `0xAE`, and also Reset (toggling GPIO pin 20) does *not* clear the GDDRAM. This makes sense for some applications; it allows quick resume from sleep without the MCU needing to resend the last screen. 

### Proposed solution:  
Zero the OLED RAM whenever we initialise the display:
- Add a clear() function to SSD1306.py that sets the first 360 bytes of GDDRAM =`0x00`. 
- Add clear() as the last step in the display.init() function.

Note: I have a working code solution and can provide a PR, if this approach is acceptable.

masonova1, 2023-01-09:
This makes sense to me. If you want to do a PR, I'll take it; Otherwise, I can implement as well. Clearing on-init seems to be the most sane option here.

ace-dent, 2023-01-09:
This solution may be sub-optimal. 
It relies on overwriting the MCU buffer then copying the buffer into the OLED GDDRAM.
There may be a situation where we want to init the OLED and keep the buffer intact (or vice versa).

masonova1, 2023-01-09:
Reopened to see if this has any way of impacting the emulator - this may not be necessary there. Are there any examples of persistent OLED garbage through restart for the emulator?

ace-dent, 2023-01-09:
+1
I like your [fix](https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/blob/82623150d21430d63b4993b46e03efedf73c700b/ThumbyGames/lib/ssd1306.py#L21-L22); short and sweet! No need to add to the emulator (I would guess it initialises with a zero filled array).

masonova1, 2023-01-09:
Sounds good. Closing for now.

commit 82623150 (2023-01-09): Clear screen on initialization

Actually fixed #52 this time
closed 2023-01-09 by masonova1