TinyCircuits/TinyCircuits-Thumby-Code-Editor #55: Update release UF2

ace-dent, 2023-01-09:
Thumby firmware contained in `ThumbyFirmware.uf2` (10-Dec-2022 ~4.2MB) is outdated (file [here](https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/blob/master/ThumbyFirmware.uf2)).
It contains old core files: Lib files 11-Jul-2022, version 1.7. `main.py` and `menu.py` seem to be ~same period. (I haven’t checked games, etc.)

Please can the UF2 be updated with latest code from this repo and games updated to match the Arcade.

ace-dent, 2023-01-18:
This is actually quite urgent for TC... I just received Thumbys with the new OLED and old firmware flashed.
When changing contrast to Low (`0` on old FW, fixed to `1` in current), I ended up with a blank screen. This may cause issues for TC customers and more support queries...

With a new UF2, it can be flashed in production.

jmtinycircuits, 2023-03-02:
This was fixed a while ago and again recently.

closed 2023-03-02 by jmtinycircuits