TinyCircuits/TinyCircuits-Thumby-Code-Editor #62: Emulator hangs on saveData.getItem and delItem

fuglaro, 2023-01-10:
Note: FAST EXECUTE works fine.

This seems to be because MicroPython is temperamental about a try/except block inside `@micropython.native`

Proposed fix: remove `@micropython.native` for getItem and delItem in ThumbyGames/lib-emulator/thumbySaves.py ONLY.

As discussed in the discord community on 10 Jan 2023:
hemlockmay — Today at 12:20 PM
I'm debugging some issues with saveData.getItem in the Emulator and I'm seeing something weird about the emulator and micropython.native. For this code:
```python
@micropython.native
def hang():
    try:
        pass
    except:
        pass

print(1)
hang()
print(2) # Never shows
```
I get this when running in FAST EXECUTE:
```
>>> 
1
2
>
```
BUT this when running in the emulator?!?!?!
```
>>> __import__('/sdfsdfsd')
1

```
Does anyone else get that behaviour in the emulator? And does `saveData.getItem` hang for anyone else when running in the emulator?

transistortester — Today at 12:37 PM
Can confirm that example hangs in the emulator

masonova1, 2023-01-10:
Working on a workaround for this -- the problem might be with exceptions themselves. Simple answer: don't use an exception in the first place.

masonova1, 2023-01-10:
We might keep the native emitter if we can get rid of the exceptions. Try these changes and see if the problem is still reproduced:
```
    # Get entry from volatile dictionary
    @micropython.native
    def getItem(self, key):
        ret = self.volatileDict.get(key, None)
        if ret is None:
            # Look for a bytes item under the key
            ret = self.volatileDict.get("__b"+key, None)
            return (None if ret is None else b64dec(ret))
        return ret
    
    # Delete entry in volatile dictionary
    @micropython.native
    def delItem(self, key):
        ret = self.volatileDict.pop(key, None)
        if ret is None:
            ret = self.volatileDict.pop("__b"+key, None)
            return (None if ret is None else b64dec(ret))
        return ret
```

masonova1, 2023-01-10:
`setName` may also not work for the same reason. I need to think about that one a little harder.

masonova1, 2023-01-10:
Closing; partially solved by my latest commit. Someone can reopen if `setName` has the same problems.

closed 2023-01-10 by masonova1

fuglaro, 2023-01-10:
Just saw your last comment. I think we are all good for setName. Thanks!