TinyCircuits/TinyCircuits-Thumby-Code-Editor #91: Emulation: Pressing the stop button breaks audio until the page is hard-reset

transistortester, 2023-04-28:
As the title says. I'm unsure if audio needs to be *actively* playing when you push the button, but doing so seems to be a surefire way to trigger the issue.
To easily reproduce the issue:
```py
import thumby
print("Playing")
thumby.audio.playBlocking(440, 3000)
print("Done")
```
Press stop as soon as you hear a tone. On subsequent runs the printed text is visible, but no tone plays.
As far as I can tell, this was introduced when the emulator started maintaining it's state across restarts.

transistortester, 2023-05-17:
From the discord (https://discord.com/channels/898292107289190461/898292174410612787/1108481635114033162):
transistortester
 — 
Today at 3:50 PM
@JasonTC I was tinkering with audio in the emulator yesterday, have you investigated this issue yet? https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/issues/91
I found that replacing the `if (this.AUDIO_BUZZER != undefined) this.AUDIO_BUZZER.stop();` with if `(this.AUDIO_BUZZER != undefined) this.AUDIO_BUZZER.frequency.value = 0;` in `emulator_wrapper.js` seems to fix the issue.

jmtinycircuits, 2023-09-28:
This should be fixed by https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/commit/3da87cb7515be1070ca5d4d4a820dafe934d61ed once the website is published again

closed 2023-09-28 by jmtinycircuits