TinyCircuits/TinyCircuits-Thumby-Code-Editor #92: Emulation: Audio accuracy - Reduce aggressiveness of exponentialRampToValueAtTime

transistortester, 2023-05-17:
From the discord (https://discord.com/channels/898292107289190461/898292174410612787/1108481714478645358)
transistortester
 — 
Today at 3:50 PM
Also somewhat related, would you consider reducing the aggressiveness of the `exponentialRampToValueAtTime`? I remember the reason for it but it makes a huge difference in the accuracy of fast sound effects - this is what it sounds like with none, 0.01, and the current 0.03 value
https://cdn.discordapp.com/attachments/898292174410612787/1108481714168279040/untitled.mp3

jmtinycircuits, 2023-09-28:
Hopefully it will sound a little better with +0.01 instead of +0.03: https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/commit/e227aab81530931c63a235003befbd89d50a8f16

Feel free to close if think that's ok.

transistortester, 2023-09-29:
0.01 should be good enough, sounds much cleaner than before. Thanks!

closed 2023-09-29 by transistortester