TinyCircuits/TinyCircuits-Thumby-Code-Editor #99: Update `Thumgeon` button code to use API

ace-dent, 2023-09-07:
As a 'core' game, Thumgeon should use the Thumby API where possible; in part for best practice/ education.
The custom button handling [code](https://github.com/TinyCircuits/TinyCircuits-Thumby-Games/blob/6203b4582a906defa2675a10499bfbc8d9fa1b40/Thumgeon/Thumgeon.py#L104-L170) should be replaced with API equivalents.
This also avoids the extra import of `Pin` from `machine`.

ace-dent, 2023-09-08:
Corresponding issue for Arcade Repo is [here](https://github.com/TinyCircuits/TinyCircuits-Thumby-Games/issues/189).

jmtinycircuits, 2023-09-28:
I removed the base games from this repository, now they're just on the Arcade https://github.com/TinyCircuits/TinyCircuits-Thumby-Games.

This and https://github.com/TinyCircuits/TinyCircuits-Thumby-Code-Editor/pull/97 shouldn't be needed here anymore.

ace-dent, 2023-09-28:
Issue is now just with the Arcade version. Closing duplicate.

closed 2023-09-28 by ace-dent