TinyCircuits/TinyCircuits-Thumby-Code-Editor #102: cannot export to pc

Buggem, 2023-12-20:
export to pc Button doesn't work (Firefox 121.0). I am a javascript dev and I have done similar things (downloading files) before and would be happy to help if you need it. Os is macOS Sonama 14.1.2 (m2)

Buggem, 2023-12-20:
Added: used js debug skills:
error is window.showSaveFilePicker is not a function

Buggem, 2023-12-20:
that is an experimental technology ye silly guys
you gotta state that

Buggem, 2023-12-20:
nice open source editor, im trying to emulate the file picker right now

Buggem, 2023-12-20:
here is some code that patches the issue, at least on firefox
also the open button aint gonna work either. it uses the same API as the [`showSaveFilePicker()`](https://developer.mozilla.org/en-US/docs/Web/API/FileSystemFileHandle/createWritable) function (the [File System API](https://developer.mozilla.org/en-US/docs/Web/API/File_System_API))
```
window.showSaveFilePicker=function(fileoptions) {
    return new Promise((resolve) => {
    resolve({
            fileoptions: fileoptions,
            kind: "file",
            createWritable: function() {
                
                return new Promise((resolve) => {
                    resolve({
                        write: function(blob){
                            var saveData = function (blob, fileName) {
                                    var a = document.createElement("a");
                                    document.body.appendChild(a);
                                    a.style = "display: none";
                                    
                                            var url = window.URL.createObjectURL(blob);
                                            a.href = url;
                                            a.download = prompt("Choose a name for the download", fileName);
                                            a.click();
                                            window.URL.revokeObjectURL(url);
                                    
                            };
                            saveData(blob, fileoptions.suggestedName)
                            resolve();
                        },
                        close:function(){}
                    })
                    //fake the fake file handles fake writable
                });
            }
        }) //fake the file handle
  });
}; 
```
I put a lot of time and effort into this ( 10 minutes) so I hope to see it implemented in your code. (im not very good)
also you must do similar for the file picker (include an invisible `<input>` instead of an invisible `<a>` as that will increase compatibility by 105%.) so yeh, be sure to at least include something like this as a fallback if the function cannot be found. Thanks, buggem (Max Parry)

Buggem, 2023-12-20:
And even weirder I can see a catch statement for the `[showOpenFilePicker()](https://developer.mozilla.org/en-US/docs/Web/API/Window/showOpenFilePicker)` call, but not the `[showSaveFilePicker()](https://developer.mozilla.org/en-US/docs/Web/API/Window/showSaveFilePicker)` call. Weird, feels like someone forgot to catch for errors there. It's lucky for me though, because it led me right to the spot.

Buggem, 2023-12-20:
for some reason i can't get the openFilePicker call to work.
```
window.showOpenFilePicker = function(fileoptions) {
	try {
    return new Promise((resolve) => {
        resolve({
            fileoptions: fileoptions,
            getFile: function() {
                let fileoptions = this.fileoptions;
                return new Promise((resolve) => {
                    let input = document.createElement('input');
                    input.type = 'file';
                    input.style = "display: none";
                    input.accept = this.fileoption.types.map(a=>a.accept).join(',')
                    input.onchange = e => {
                        resolve(e.target.files[0]);
                    }
                    input.click();
                    window.myFileInput = input; // for debugging
                }); // our part is done, we can let the file class do the rest... woo-hoo
            }
        })
    });
	} catch(e) { console.log(e.message); }
}
```
my code so far

Buggem, 2023-12-20:
finished and working:
```
window.showOpenFilePicker = function(fileoptions) {
    console.log("here")
    window.openfileopt = fileoptions;
    return new Promise((resolve) => {
        
        alert("inside promise openfilepicker")
        resolve([{
            
            getFile: function() {
                alert("got to getfile")
                
                return new Promise((resolve) => {
                    let input = document.createElement('input');
                    input.type = 'file';
                    input.style = "display: none";
                    input.onchange = e => {
                        resolve(e.target.files[0]);
                    }
                    input.click();
                    window.myFileInput = input; // for debugging
                }); // our part is done, we can let the file class do the rest... woo-hoo
            }
        }])
    });
    
}
```

Buggem, 2023-12-20:
could you implement my code? will open a pull request if you want me to

Buggem, 2023-12-22:
congratulations, you changed your script and broke my code :)
working on a fix, if you do this again ill have to create a repo for this ;) lol

Buggem, 2023-12-22:
umm no you didnt sorry

it was just micropython code instead of blockly code ;0

Buggem, 2023-12-22:
https://github.com/Buggem/sysstorapi-hack/tree/main

incase of future updates, here it is

Buggem, 2023-12-31:
hurry up and fix

Buggem, 2024-04-06:
this is just useless irl

Buggem, 2024-04-06:
been 4 months bro

ace-dent, 2024-04-06:
Please can you raise a PR for review?

Buggem, 2024-04-06:
@ace-dent yes i can, and i will try